strange-v/MHZ19 #1: HardwareSerial1 in Arduino Leonardo

khseal, 2017-11-23:
Hi want to use the second port in arduino leonardo.
His name is serial1.
I do not understand how to initialize it correctly.

strange-v, 2017-11-23:
Use
`MHZ19 mhz(&Serial1);`

I've added "sw_get_values" example.

khseal, 2017-11-23:
I tried this option. After that, arduino became an unknown device .. =)

strange-v, 2017-11-23:
It's quite strange.
Unfortunately I don't have Arduino Leonardo, but I've used this lib with hardware serial on Arduino Mega without any issues.
Probably you could check Serial1 with FTDI adapter to be sure that it works.

khseal, 2017-11-23:
It works exactly with the other code, and with your library this is the problem = (

strange-v, 2017-11-23:
Please provide the complete sketch/code which you use to reproduce the problem.

khseal, 2017-11-23:
[image]

strange-v, 2017-11-23:
There was a typo in the data reading, fixed.
Also interfaces was updated to be less confusing (see examples).

commit 90405bf3 (2017-11-23): Fix #1
Updated interface to be less confusing
Updated examples
closed 2017-11-23 by strange-v