strange-v/MHZ19 #5: Thank you for the good information. I have a question.

rimmomo, 2019-05-28:
I'm using my hardware. (ESP32)
How do I set the range?

#include <MHZ19.h>
HardwareSerial mhz19(1);
MHZ19 mhz(&mhz19);

void setup() {
     Serial.begin(9600);
     delay(800);
     mhz19.begin(9600);

     mhz.setAutoCalibration(false);
     mhz.setRange(3);
}

void main() {
    MHZ19_RESULT response = mhz.retrieveData();
    if (response == MHZ19_RESULT_OK)
    {
       Serial.print(F("CO2: "));
       Serial.println(mhz.getCO2());
       Serial.print(F("Temperature: "));
       Serial.println(mhz.getTemperature());
       Serial.print(F("Accuracy: "));
       Serial.println(mhz.getAccuracy());
    }
    else
    {
       Serial.print(F("Error, code: "));
       Serial.println(response);
    }
}

I'm a beginner, so I don't know.
Thank you for the good information.

strange-v, 2019-05-28:
What range value do you want to set?
In any case, you have to use enum defined in MHZ19.h, e.g. for range 3000 ppm code will be like this:
`mhz.setRange(MHZ19_RANGE_3000);`

This is the list of valid ranges:
MHZ19_RANGE_1000,
MHZ19_RANGE_2000,
MHZ19_RANGE_3000,
MHZ19_RANGE_5000,
MHZ19_RANGE_10000

rimmomo, 2019-05-29:
> What range value do you want to set?
> In any case, you have to use enum defined in MHZ19.h, e.g. for range 3000 ppm code will be like this:
> `mhz.setRange(MHZ19_RANGE_3000);`
> 
> This is the list of valid ranges:
> MHZ19_RANGE_1000,
> MHZ19_RANGE_2000,
> MHZ19_RANGE_3000,
> MHZ19_RANGE_5000,
> MHZ19_RANGE_10000

Thank you!

closed 2019-05-30 by strange-v