strange-v/MHZ19 #6: I can not use in ESP8266

HanSeoHyun, 2019-06-03:
Hello, I tested your PWM code with UNO to MHZ19B. It works very well.
So I want use your library with nodeMCU ESP8266 12E. But It does not work.

my code:
#include <MHZ19PWM.h>   

MHZ19PWM mhz(5, MHZ_DELAYED_MODE);

void setup()
{
  Serial.begin(115200);          
  Serial.println(F("Starting..."));
  mhz.useLimit(5000);
}

void showValue()
{
  unsigned long start = millis();
  float co2 = mhz.getCO2();
  unsigned long duration = millis() - start;

  Serial.print(F("CO2: "));       
  Serial.println(co2);  
  Serial.print(F("Duration: "));
  Serial.println(duration);        
  Serial.println();
}

void loop()
{
  showValue();        
  delay(5000);          
}

my direction:
nodeMCU ---- sensor
Vin ---- Vin
GND ---- GND
D1(GPIO5) ---- PWM

my serial monitor:
[image]

help me please....

strange-v, 2019-06-03:
Hello.
Seems a lot of things was changed in the Arduino core for esp8266 and I wouldn't say it become more stable.
If you are in harry the easiest way is to use an older core, e.g. 2.1.0 (I've just tested - it works).

I will look at this issue but not sooner than the next weekend.

HanSeoHyun, 2019-06-03:
Thank you for your comment.

I changed only esp8266 core version like below image. Is it right?
[image]

but it does not work well too...
[image]

Do you think I need to modify something else?

strange-v, 2019-06-04:
No, everything looks okay.
UART is tested, but you have to take into account that Arduino core for ESP8266 version 2.4 and later has troubles with Software Serial (check https://github.com/plerup/espsoftwareserial/issues/63).
So you have to use version 2.3.

HanSeoHyun, 2019-06-04:
I try to nodeMCU on UART ESP8266 version 2.3, But It does not work.
Actually, my UNO on UART mode does not work same results.
They both print in serial monitor: "Error, code: 5" (infinitely)
Because I try to another board nodeMCU.

So I can test only UNO on PWM...

strange-v, 2019-06-04:
Error code 5 equals MHZ19_RESULT_ERR_TIMEOUT. The sensor didn't respond for more than 500 milliseconds.
Please show/describe your wairing and what code is used.

HanSeoHyun, 2019-06-04:
I try to UNO on UART mode
my serial monitor with code:
[image]

my connection
MH_Z19B ----- Arduino UNO
Rx ------ 3
Tx ------ 2
GND ---- GND
Vin ----- 5V

this time to test, "Error, code: 5" is not printed infinitely. Sometimes Co2 checked 390~400, but I don't believe this values are real.

strange-v, 2019-06-04:
I've no idea how I can help you. The MHZ19 library works without any issues on Arduino Nano (the same ATmega328P that in Uno), just tested it again https://i.imgur.com/zL59nmG.jpg it also works on Mega, esp8266 and esp32 (tested a few weeks ago).

Maybe you have some issues with the sensor itself.

P.S.
The MHZ19PWM library currently has issues on esp8266 (at least), probably I'll move it to a separate repository/library.

HanSeoHyun, 2019-06-05:
I am really thank you about your helps. 
My goal was to connect two sensors to one boards(nodeMCU). 
But I could not do it, So I had no choice. 

There are my finally test lists:
UNO - UART ( X )
UNO - PWM ( O ) 
nodeMCU esp8266 - UART ( X )
nodeMCU esp8266 - PWM ( X )

I tried many tests, then UNO - PWM was works good. maybe, my test was something wrong in yesterday.
Finally I use two board that is UNO, nodeMCU.
It was very inefficient, But It was best choice for me... 

Thank you so much for your replies.

strange-v, 2019-06-06:
I've used four sensors simultaneously on Arduino Mega (three via hardware serial + one via software serial) to compare results. You could try esp32 it has one additional hardware serial and supports software one too (AFAIK).

I'm glad you managed to solve your task.

closed 2019-06-07 by strange-v

jwktje, 2019-12-04:
I have this issue. I want to use the library on a Wemos D1 Mini using ports D1 and D2 for software serial. I get Error 3. Any tips on this?

strange-v, 2019-12-04:
@jwktje this issue was more about the PWM library.
You are trying to use SoftwareSerial on ESP8266 it may work or may not due to this [bug](https://github.com/plerup/espsoftwareserial/issues/63). In general, I would recommend using ESP32 instead of ESP8266 in all new projects (at least if they would use Serial communication).

If you think that the MHZ19 library has some bug - please open a new issue.