strange-v/MHZ19 #7: How to calibrate?

alxpy, 2019-06-18:
Hello! Thanks for the great library!

Could you prompt how to calibrate sensor using your library?

Is that all I need to do (of course with more than 20 minutes of clean air)?
```cpp
#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(10, 11);
MHZ19 mhz(&ss);

void setup()
{
  ss.begin(9600);
  mhz.calibrateZero();
}

void loop() {}
```

strange-v, 2019-06-19:
Hello.
Unfortunately, you haven't specified what sensor you are going to calibrate.

As far as I remember MH-Z19' zero point calibration is really ZERO points. In contrast, a datasheet for MH-Z19B says: "ZERO POINT is 400PPM, PLS MAKE SURE THE SENSOR HAD BEEN WORKED UNDER 400PPM FOR OVER 20MINUTES".

So the answer is yes, the code above sends zero point calibration command to the sensor but depending on the sensor you will get a different result.

alxpy, 2019-06-19:
I uploaded this code to arduino nano with connected MH-Z19B and wait 20 minutes near by open window. But it did not help.
Hm, maybe I have different problem (bad power by example).

Thank!

closed 2019-06-19 by alxpy