strange-v/MHZ19 #8: Pins used on Wemos D1 mini (ESP8266)

quintendewilde, 2020-10-10:
Hello, 

What pins need to be used to get the ppm value of the MHZ19 on a Wemos D1 mini
an alternative ESP8266 board. 
I'm not sure how to wire this or do I just use the RX TX standard pins on the wemos?

strange-v, 2020-10-11:
Hello, 
ESP8266 is a bad choice for this purpose. It has only one hardware serial and SoftwareSerial lib doesn't work correctly from some release (do not remember the exact version).
You can use RX TX, but in this case, you should not use it for debugging.
ESP32 has more hardware serial interfaces.

quintendewilde, 2020-10-11:
And do you have a wiring example for the lolin32 and is there a sketch for lolin32 (I haven't checked myself, I'm currently trying to find my lolin32 to try it out. ) Thanks for the explanation so far!

strange-v, 2020-10-11:
Use hw_get_values example.
Hardware serial on ESP32 can be removed to almost any pin
`Serial1.begin(115200, SERIAL_8N1, rxPin, txPin);`

quintendewilde, 2020-10-11:
@strange-v MHZ19 requires 5v right. So I'll have to use a level shifter.. Or did you test this with 3v?

strange-v, 2020-10-11:
It requires from 3.6 to 5.5V, but the logic pins use 3.3V as a high level (at least MH-Z19B). Check the dataset for your sensor.

strange-v, 2020-10-18:
@quintendewilde have you managed to use lib on esp8266?

quintendewilde, 2020-10-18:
No sorry I didn't figure it out. I used another library that changed the
serial to other ports and it works in that I get the ppm but not
temperature...

Interested in it I can give you the link of the github when I'm home.

Our LiB worked better on the Arduino though!

Regards Q

strange-v, 2020-10-19:
@quintendewilde good to now, that you managed to use your sensor.

FYI, I've just tested on ESP8266 (NodeMCU ESP-12e), and everything works fine, it seems that SoftwareSerial has been improved/fixed.

```cpp
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(13, 12); //rx, tx (D7, D6) on NodeMCU ESP-12e
MHZ19 mhz(&ss);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  ss.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Min CO2: "));
    Serial.println(mhz.getMinCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}
```

I think this issue can be closed.

quintendewilde, 2020-10-30:
Hi I went back for your library after all. Everything but accuracy and CO2 min works. 
I have no clue where to attach the PWM pin or is this not necessary?

strange-v, 2020-10-31:
What are you getting as a result of getAccuracy()?

Have you seen this comment in the header file?
```cpp
// It isn't ready to use minimum CO2 value, additional calculations have to be applied
int getMinCO2();
```

PWM pin is not needed for MHZ19 library. It can be used with MHZ19PWM to read CO2 only, but this library has issues with the latest esp8266/esp32 releases.

quintendewilde, 2020-10-31:
Oops I didn't read that comment from the code. 

The getAccuracy() gives me 0

strange-v, 2020-10-31:
> The getAccuracy() gives me 0

I would recommend monitoring (log to some file, database, whatever) this value for 24h or more, it will clarify whether it is changing or not.

In general, accuracy and CO2 min values aren't described in the sensor datasheet and were deducted by enthusiasts, so I wouldn't expect much here.

strange-v, 2021-01-13:
Closed due to inactivity.

closed 2021-01-13 by strange-v