strange-v/MHZ19 #10: invalid conversion from 'HardwareSerial*' to 'int' [-fpermissive]

quintendewilde, 2020-10-12:
@strange-v Hi I'm trying to get this to work on an ESP32 but because I thought or think my sensor is defected. 
I'm trying to get this to work on an Arduino Mega to cancel or not cancel that out. 

When trying SW or HW example I get a similar error: 

`Arduino: 1.8.13 (Mac OS X), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new can abort), All SSL ciphers (most compatible), 4MB (FS:none OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

sketch_oct12a:3:19: error: invalid conversion from 'HardwareSerial*' to 'int' [-fpermissive]
 MHZ19 mhz(&Serial1);
                   ^
In file included from /Users/quintendewilde/Documents/Arduino/sketch_oct12a/sketch_oct12a.ino:1:0:
/Users/quintendewilde/Documents/Arduino/libraries/MHZ19-master/src/MHZ19.h:45:2: error:   initializing argument 1 of 'MHZ19::MHZ19(int)' [-fpermissive]
  MHZ19(int pwm);
  ^
/Users/quintendewilde/Documents/Arduino/sketch_oct12a/sketch_oct12a.ino: In function 'void loop()':
sketch_oct12a:15:3: error: 'MHZ19_RESULT' was not declared in this scope
   MHZ19_RESULT response = mhz.retrieveData();
   ^
sketch_oct12a:15:16: error: expected ';' before 'response'
   MHZ19_RESULT response = mhz.retrieveData();
                ^
sketch_oct12a:16:7: error: 'response' was not declared in this scope
   if (response == MHZ19_RESULT_OK)
       ^
sketch_oct12a:16:19: error: 'MHZ19_RESULT_OK' was not declared in this scope
   if (response == MHZ19_RESULT_OK)
                   ^
sketch_oct12a:19:24: error: 'class MHZ19' has no member named 'getCO2'
     Serial.println(mhz.getCO2());
                        ^
sketch_oct12a:21:24: error: 'class MHZ19' has no member named 'getTemperature'
     Serial.println(mhz.getTemperature());
                        ^
sketch_oct12a:23:24: error: 'class MHZ19' has no member named 'getAccuracy'
     Serial.println(mhz.getAccuracy());
                        ^
exit status 1
invalid conversion from 'HardwareSerial*' to 'int' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

PIN
TX  to RX 19 on mega
RX to TX 18 on mega

strange-v, 2020-10-12:
Hi @quintendewilde,
You sed you are using Arduino Mega (it is AVR based board, ATmega2560) but in your config I see "LOLIN(WEMOS) D1 R2 & mini". WeMos D1 R2 is an ESP8266 based board that has similar to the Arduino Mega form factor.
So what MCU are you using?
Please show your sketch too.

quintendewilde, 2020-10-12:
So sorry, It's getting late and I'm too eager to get this to work. While it isn't working at all. 

Indeed I forgot to change the port. 

I did that and then I get the following message with this sketch. 

`#include <MHZ19.h>

MHZ19 mhz(&Serial1);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  Serial1.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}`

ERROR: 

`Arduino: 1.8.13 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/Users/quintendewilde/Documents/Arduino/sketch_oct12a/sketch_oct12a.ino: In function 'void loop()':
sketch_oct12a:15:3: error: 'MHZ19_RESULT' was not declared in this scope
   MHZ19_RESULT response = mhz.retrieveData();
   ^~~~~~~~~~~~
sketch_oct12a:16:7: error: 'response' was not declared in this scope
   if (response == MHZ19_RESULT_OK)
       ^~~~~~~~
/Users/quintendewilde/Documents/Arduino/sketch_oct12a/sketch_oct12a.ino:16:7: note: suggested alternative: 'remove'
   if (response == MHZ19_RESULT_OK)
       ^~~~~~~~
       remove
sketch_oct12a:16:19: error: 'MHZ19_RESULT_OK' was not declared in this scope
   if (response == MHZ19_RESULT_OK)
                   ^~~~~~~~~~~~~~~
sketch_oct12a:19:24: error: 'class MHZ19' has no member named 'getCO2'; did you mean 'getppm'?
     Serial.println(mhz.getCO2());
                        ^~~~~~
                        getppm
sketch_oct12a:21:24: error: 'class MHZ19' has no member named 'getTemperature'
     Serial.println(mhz.getTemperature());
                        ^~~~~~~~~~~~~~
sketch_oct12a:23:24: error: 'class MHZ19' has no member named 'getAccuracy'
     Serial.println(mhz.getAccuracy());
                        ^~~~~~~~~~~
exit status 1
'MHZ19_RESULT' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

strange-v, 2020-10-12:
Interesting.
Cannot reproduce it using Arduino: 1.8.13 on Windows (compilation completed successfully for Arduino Mega board).
Let me think for a while.

strange-v, 2020-10-12:
@quintendewilde, I've just committed new changes, try to redownload the library.
I've been using @pragma once instead of include guards in this library. I don't think this is the reason, but let's try.

quintendewilde, 2020-10-12:
@strange-v It works!!! 👍  :D

Now getting this to work on a Wemos D1 mini is my next challenge! 

Thanks for the help!!! I know the sensor is now working and not faulty!

`CO2: 1096
Temperature: 25
Accuracy: 0
CO2: 1095
Temperature: 25
Accuracy: 0
CO2: 1026
Temperature: 25
Accuracy: 0
CO2: 960
Temperature: 25
Accuracy: 0
CO2: 872
Temperature: 25
Accuracy: 0
`

Is the output! I guess it will go on lower to 400 If I read most of the articles correctly.

strange-v, 2020-10-12:
@quintendewilde good to hear!
400ppm is the lowest value (411.29 [for now](https://www.co2.earth/)), it'll be higher if you sit nearby.

You could try using SoftwareSerial on esp8266, maybe they fixed the issue in the latest releases, I haven't tested so far.

strange-v, 2020-10-13:
Fixed in the commit 09b424a419347b274cbc29c42c8c566886a372b5

closed 2020-10-13 by strange-v