strange-v/MHZ19 #12: error code:5

vignesh8458, 2021-01-24:
when i run hw_get_values.ino

i am getting error code:5

Please help

connection:

MH-Z19b      |       Esp8266 
 
Vin                |            vin
GND              |        GND
Tx                  |          Rx
Rx                 |           Tx

strange-v, 2021-01-24:
Error code 5 represents communication timeout (MHZ19_RESULT_ERR_TIMEOUT).

So, you are using hardware serial to communicate with the sensor, what you are using for the debug? ESP8266 has only one HW serial.

vignesh8458, 2021-01-24:
i don't know, how to use debug?

vignesh8458, 2021-01-24:
I am new to IoT

what is the best way to get data from the MH-Z19?

strange-v, 2021-01-24:
You've just said, that you got code: 5 (it is a debug info).
Could you post the whole code here?

Use software serial for esp8266 or use esp32.

vignesh8458, 2021-01-24:
Code: 

#include <MHZ19.h>

MHZ19 mhz(&Serial1);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  Serial1.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}

vignesh8458, 2021-01-24:
could you please share the code for esp8266?

strange-v, 2021-01-24:
Serial1 on esp8266 has only TX capabilities.
Use sw_get_values.ino example.

vignesh8458, 2021-01-24:
Thanks 

i will try and update to you.

vignesh8458, 2021-01-24:
I have tried in new Esp8266, I got this below error.

Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

IROM   : 237596          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 28792   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1248  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 736   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 25328 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 268372 bytes (25%) of program storage space. Maximum is 1044464 bytes.

Global variables use 27312 bytes (33%) of dynamic memory, leaving 54608 bytes for local variables. Maximum is 81920 bytes.

esptool.py v2.8

Serial port COM3

Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):

  File "C:\Users\Vignesh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/Vignesh/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main

    esp.connect(args.before)

  File "C:/Users/Vignesh/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect

    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

vignesh8458, 2021-01-24:
I am using Esp8266 Amica nodemcu.

vignesh8458, 2021-01-24:
I have uploaded the sw_get_values.ino.

again I got error code: 5

connection:

MH-Z19b | Esp8266

Vin | Vin
GND | GND
Tx | Rx
Rx | Tx

strange-v, 2021-01-24:
Did you connect it like this?

MH-Z19b | Esp8266
Vin | Vin
GND | GND
Tx | 17
Rx | 16

vignesh8458, 2021-01-24:
I have connected like this:

MH-Z19b | Esp8266

Vin | Vin
GND | GND
Tx | Rx
Rx | Tx

Could you please mark the connection on the below image(esp8266?

[image]

strange-v, 2021-01-24:
Ok, replace in the code
`SoftwareSerial ss(17,16);`
with
`SoftwareSerial ss(5,4);`

then connect
MH-Z19b | Esp8266
Vin | Vin
GND | GND
Tx | GPIO5
Rx | GPIO4
and try.

With software serial, you can use almost any pins.

vignesh8458, 2021-01-24:
I think I am getting output,

But min co2 and Accuracy is 0?    (i hope this is an issue)

And can i change the delay time 15000?

 
Output:

CO2: 862
Min CO2: 0
Temperature: 30
Accuracy: 0

vignesh8458, 2021-01-24:
Code: 

#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(5,4);
MHZ19 mhz(&ss);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  ss.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Min CO2: "));
    Serial.println(mhz.getMinCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}

strange-v, 2021-01-24:
Cool.
Min CO2, accuracy and even temperature aren't described in the sensor's datasheet and were deducted by enthusiasts, so I wouldn't expect much here.

vignesh8458, 2021-01-24:
I hope the CO2 output is the correct one?

strange-v, 2021-01-24:
Yes, but you should know that this sensor performs auto-calibration by the lowest detected value once per 24 hours.

vignesh8458, 2021-01-24:
how do I disable the auto-calibration?

strange-v, 2021-01-24:
Add this line after ss.begin(9600); in the setup:
`mhz.setAutoCalibration(false);`

vignesh8458, 2021-01-24:
thanks 

what is your full name?

vignesh8458, 2021-01-25:
hi ,

Can I use other pins to getting sensor data, because I am using i2c display16*2?

code:

SoftwareSerial ss(5,4);

to

SoftwareSerial ss(  );

strange-v, 2021-01-25:
As I previously said
> With software serial, you can use almost any pins.

vignesh8458, 2021-01-27:
When i add this line (  mhz.setAutoCalibration(false);   )

I am getting a co2 level output  is 3  ( co2: 3 )

Please help me with this.

Code:

#include <SoftwareSerial.h>
#include <MHZ19.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <SimpleTimer.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
char auth[] = "FE6RisArcNKdGmfmCelk3pWsPRCtfnMs";

SoftwareSerial ss(5,4);
MHZ19 mhz(&ss);

/* WiFi credentials */
char ssid[] = "camera";
char pass[] = "falcon$1947";

SimpleTimer timer;

int data = 0;
int LED = 12;
int lcdColumns = 16;
int lcdRows = 2;
LiquidCrystal_I2C lcd(0x27, lcdColumns, lcdRows); 

void setup() {
// put your setup code here, to run once:
Serial.begin(115200);
Serial.println(F("Starting..."));
Blynk.begin(auth, ssid, pass);
pinMode(LED, OUTPUT);
timer.setInterval(1000L, getSendData);
ss.begin(9600);
mhz.setAutoCalibration(false);

}

void loop() {

    MHZ19_RESULT response = mhz.retrieveData();
    if (response == MHZ19_RESULT_OK)
    Wire.begin(2,0);
  // initialize LCD
  lcd.init();
  // turn on LCD backlight                      
  lcd.backlight();
  lcd.print("CO2:       ");
  lcd.println(mhz.getCO2());

  if (1<5);
{
// put your main code here, to run repeatedly:
    Serial.print(F("CO2:  "));
    Serial.println(mhz.getCO2());

timer.run(); // Initiates SimpleTimer
Blynk.run();

}
if (response == MHZ19_RESULT_OK)
if (mhz.getCO2()>800)
  {

digitalWrite(LED, HIGH); // turn the LED on
delay(1000); // wait for a second
tone(14, 780, 180);
    delay(1000);
digitalWrite(LED, LOW); // turn the LED off
//delay(1000); // wait for a second
  }
  
  else 
{
  digitalWrite(LED, LOW);
  noTone(0);
}
delay(5000);

}

/***************************************************
 * Send Sensor data to Blynk
 **************************************************/
void getSendData()
{
   
  Blynk.virtualWrite(V2, mhz.getCO2()); //virtual pin V3
}

strange-v, 2021-01-27:
1. Use a code tag
2. Always post as a little example as possible, that reproduces the issue.

Are you saying that your code works okay without adding `mhz.setAutoCalibration(false);` and shows incorrect CO2 just after adding this line?

vignesh8458, 2021-01-27:
Yes,

Continuously I am getting CO2 output: 3

strange-v, 2021-01-27:
Ok, I'll try to reproduce.

strange-v, 2021-01-28:
Interesting, it works on my setup.
According to the datasheet, the sensor shouldn't respond to the ABC logic on/off command, but maybe something was changed in the latest revisions. So try this:
```
ss.begin(9600);
mhz.setAutoCalibration(false);
// wait for the response and read it
delay(10);
while (ss.available() > 0)
{
  ss.read();
}
```

strange-v, 2021-04-20:
Closed due to inactivity

closed 2021-04-20 by strange-v