strange-v/MHZ19 #13: Can you add in your library Preheating and turning OFF Auto Calibration funtions ?

MIKHANYA, 2021-06-22:


strange-v, 2021-06-23:
Hi @MIKHANYA,

You can turn off auto-calibration via setAutoCalibration, e.g:
`mhz.setAutoCalibration(false);`

full example:
```cpp
#include <MHZ19.h>

MHZ19 mhz(&Serial1);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  Serial1.begin(9600);
  mhz.setAutoCalibration(false);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}
```

What is preheating? Do not remember such functionality in this sensor.

MIKHANYA, 2021-06-23:
Thanks for the example, "Preheating" I saw this function in other libs. I don't actually know how it works there, but I think it's like a filter. When we send a request for data when the sensor just plugged in(I use mh-z19b), at first there is an error, then it sends a lower threshold of CO2 410 ppm. After a period of time, in my case 70 sec, it shows true CO2lvl. Maybe the time period when it sends 410 ppm may be different and depend on the temperature outside.

strange-v, 2021-06-23:
Okay, I see.
I understand the benefits of such functionality though, I'd like to keep the code simple and straightforward. The mentioned logic can be implemented in a different layer (in a user code) if it is required for the particular use case.

closed 2021-07-19 by strange-v