strange-v/MHZ19 #14: Temperature not showing correctly on MH-Z19C

vignesh8458, 2021-09-07:
Hi,

I have used the same code on mhz-19b it's working.

In mh-z19c temperature value showing continuously -40

I am getting this error(temperature:-40):

Min CO2: 0
Temperature: -40
Accuracy: 0
CO2: 1031
Min CO2: 0
Temperature: -40
Accuracy: 0
CO2: 1030
Min CO2: 0
Temperature: -40
Accuracy: 0
CO2: 1030
Min CO2: 0
Temperature: -40
Accuracy: 0

Connection:

MH-Z19C      |      ESP -12E

Vin                 |      5V
GND              |      GND
Tx                   |      D1
Rx                  |      D2

CODE:

#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(5,4);
MHZ19 mhz(&ss);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("Starting..."));

  ss.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Min CO2: "));
    Serial.println(mhz.getMinCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(15000);
}

strange-v, 2021-09-07:
Hi,

I haven't heard about MH-Z19C till today, so have no experience using it. It is a new sensor with a new firmware that is not fully compatible with the library.

According to the MH-Z19C datasheet, it uses the same command to init reading and the same output format for CO2 value. All other values are not mentioned there (for MH-Z19B they were deducted by enthusiasts) also, the datasheet contains only one additional command
`0x79- On/Off Self-calibration for Zero Point`
in MH-Z19B `0x79` was used to turn ABC logic on/off.

vignesh8458, 2021-09-07:
Do you suggest any code for the temperature issues?

strange-v, 2021-09-13:
This is not an issue. You are using a new sensor that may not have temperature data (at least in the same format).
Unfortunately, cannot help with this.

strange-v, 2021-09-19:
Closed due to inactivity.

closed 2021-09-19 by strange-v

vignesh8458, 2021-09-19:
Thanks for the clarification.