strange-v/MHZ19 #15: Error code 3 and 4

lata22, 2021-12-17:
Hi, I´m trying to get this sensor working but unfourtunately im not able.

I´m using MH-Z19B 0-5000ppm sensor and ESP8266 Nodemcu v3
I´ve connected the MH-Z19B tx pin to ESP8266 D1 pin and MH-Z19B rx pin to ESP8266 D2 pin and MH-Z19B Vin to the positive of a 5v 2A source and the MH-Z19B Gnd to the ground of the same power source.

And I´m using this code:

#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(D4,D5); //Maybe the problem is in this constructor that I´m passing the wrong pin numbers? I´ve already tried withoud the 'D'
MHZ19 mhz(&ss);

void setup()
{
  Serial.begin(9600);
  Serial.println(F("Starting..."));

  ss.begin(9600);
}

void loop()
{
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Min CO2: "));
    Serial.println(mhz.getMinCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }
  
  delay(1000);
}

And I´m getting:
Error, code: 3 and sometimes Error, code: 4

What am I doing wrong?

Thanks!!

lata22, 2021-12-17:
Well now with this code:
#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(5,4);
MHZ19 mhz(&ss);

void setup()
{
Serial.begin(115200);
Serial.println(F("Starting..."));

ss.begin(9600);
}

void loop()
{
  while (ss.available() > 0)
  {
    Serial.print(".");
    ss.read();
  }
  
  MHZ19_RESULT response = mhz.retrieveData();
  if (response == MHZ19_RESULT_OK)
  {
    Serial.print(F("CO2: "));
    Serial.println(mhz.getCO2());
    Serial.print(F("Min CO2: "));
    Serial.println(mhz.getMinCO2());
    Serial.print(F("Temperature: "));
    Serial.println(mhz.getTemperature());
    Serial.print(F("Accuracy: "));
    Serial.println(mhz.getAccuracy());
  }
  else
  {
    Serial.print(F("Error, code: "));
    Serial.println(response);
  }

delay(10000);
}

I´m getting values but I don´t know how to convert it to ppm.
I´m getting:
11:05:07.833 -> CO2: 2142
11:05:07.833 -> Min CO2: 0
11:05:07.833 -> Temperature: 27
11:05:07.833 -> Accuracy: 0

How do I convert the CO2 2142 value to ppm? This number can´t be in ppm because I would rather be dead with this amount of co2

strange-v, 2021-12-18:
It is already in ppm.

strange-v, 2021-12-20:
Sorry, didn't have time to answer your question in a more detailed manner.
A few additional points:
- it takes 24 hours for the sensor to re-calibrate itself, so leave it connected and wait (and open a window for a few minutes)
- 2142 ppm isn't as high as you think

> 2,000-5,000 ppm | Headaches, sleepiness and stagnant, stale,  stuffy air. Poor concentration, loss of attention, increased heart rate and slight nausea may also be present.

strange-v, 2021-12-28:
Closed due to inactivity.

closed 2021-12-28 by strange-v