strange-v/MHZ19 #16: Guru Meditation Error: Core 1 panic'ed (IllegalInstruction)

BelKed, 2022-05-02:
I downloaded the library from the `master` branch and after uploading the example code (`hw_get_values.ino`) without any changes to ESP32 it doesn't seem to work:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Starting...
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d0dc0: 54544547 45777464 56644574
Core 1 register dump:
PC      : 0x400d0dc4  PS      : 0x00060330  A0      : 0x800d1630  A1      : 0x3ffb1f90  
A2      : 0x0800001c  A3      : 0x3ffbfe48  A4      : 0x3ffbfeb4  A5      : 0x80000020  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0c3f  A9      : 0x3ffb1f40  
A10     : 0x3ffbfe48  A11     : 0x0000000a  A12     : 0x0800001c  A13     : 0x00000009  
A14     : 0x00000009  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0dc4:0x3ffb1f90 0x400d162d:0x3ffb1fb0 0x400860ed:0x3ffb1fd0

Rebooting...
```

---

After placing the stack trace in the ESP exception decoder, the following is displayed:

```
PC: 0x400d0dc4: MHZ19::retrieveData() at /Users/user/Documents/Arduino/libraries/MHZ19/MHZ19.cpp line 98
EXCVADDR: 0x00000000

Decoding stack results
0x400d0dc4: MHZ19::retrieveData() at /Users/user/Documents/Arduino/libraries/MHZ19/MHZ19.cpp line 98
0x400d162d: loopTask(void*) at /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 23
0x400860ed: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

---

If I can provide something else related to this issue, feel free to write...

strange-v, 2022-05-03:
I see that framework is updated (a new major version was released), maybe something significant was changed in the API, but it is weird, as the lib only uses hardware serial.
I'll try to run the HW example on my end.

strange-v, 2022-05-03:
Have you tried Serial2?
AFAIK, by default Serial1 is mapped to IO9 and IO10 which are used for flash.
So either remap Serial1 to different pins or use Serial2.

I've just tested and it works on the latest framework.

BelKed, 2022-05-04:
It seems to be working with `Serial2`. I know that `Serial1` has certain limitations, but actually, I didn't really realize that I can't use it...
Thanks for pointing me to the right direction :)

closed 2022-05-04 by BelKed

strange-v, 2022-05-05:
Nice to hear that it works.

You can use Serial1, just remap pins like this:
```cpp
Serial1.begin(9600, SERIAL_8N1, 16, 17); //rx, tx
```