strange-v/MHZ19 #18: MH-Z19B: Very high values and 0 accuracy

Michal-Szczepaniak, 2024-09-19:
Hi, I've used your library but I'm having issue with the sensor and I'm hoping you could shine some light on the issue. When I first connected the sensor, library seems to be communicating without errors, and in like first minute it reported 500ppm then it jumped to 5000ppm, assuming 5000 is the limit by default. 

After some time it started going down seemingly with change in temperature  but its still hovering around 4000 even 24h later. I've enabled auto-calibration and in documentation it stated the auto-calibration cycle is 24h so i was hoping after 24h it will change the accuracy and maybe lower the values to something more sane but no such thing happened the accuracy is still 0 and values are still very high. 

I have my sensor connected to the esp32 if that helps.

strange-v, 2024-09-20:
Hi. I haven't been working with those sensors for a while. Make sure that it is running 24 hours continuously, each reconnect will cause the reset of the ABC timer. If you are using version "B" of the sensor, do a manual calibration, it'll set the current value to 400 ppm.

Michal-Szczepaniak, 2024-09-20:
Manual calibration is kinda impossible because i would have to get the sensor outside  and back without losing power no?

version B cannot perform autocalibration?

Also I've figured out i had problem on my part, and i connected 3.3V to sensor's vin not 5V so values are more sane now (700 instead of 4000) but accuracy still 0 :/

strange-v, 2024-09-20:
Open the window, close the room, and leave for 30 minutes, something like this.
Version B cannot do autocalibration. The original, not B version cannot do manual calibration without the lab, as it calibrates to 0 instead of 400.

Ignore accuracy, it is an internal thing.

Michal-Szczepaniak, 2024-09-20:
But the datasheet says it can so like what the hell? https://www.winsen-sensor.com/d/files/infrared-gas-sensor/mh-z19b-co2-ver1_0.pdf

But even without calibration, when i opened window and left the room the value went down to almost 400ppm so i think i don't need much calibration already… and its not like scientific scenario anyway

ignore accuracy, got it. I think it's working fine honestly so i'll just close this and sorry for bothering!

closed 2024-09-20 by Michal-Szczepaniak

strange-v, 2024-09-20:
Sorry, typo... Version B **_can_** do autocalibration.

Michal-Szczepaniak, 2024-09-20:
yeah so i just enabled autocalibration and it seems to show correct values already so i'm happy :)

thanks for the lib!