strange-v/MHZ19 #19: Error Code5

quintendewilde, 2024-10-23:
I'm getting this error 5 when (re)connecting my MHZ19B for quite a while now.

1. I saw the other issue and looked at the steps!
2. I have it wired to my Wemos D1 mini (D7 (13) RX, D8 (15) TX).
3. Serial baudrate is correct.
4. I said a delay(2000) after ss.begin(9600)
5. 5v from the Wemos is powering the MHZ (I see the light blinking)

This is the code:

`#include <SoftwareSerial.h>
#include <MHZ19.h>

SoftwareSerial ss(13, 15); // RX, TX (D7, D8)
MHZ19 mhz(&ss);

const int ledPin = D0; // LED pin
const unsigned long intervalMhz = 10000; // Sensor interval (10 seconds)
unsigned long previousMillisMhz = 0;

void setup() {
    Serial.begin(115200);
    ss.begin(9600);
    pinMode(ledPin, OUTPUT);
    digitalWrite(ledPin, LOW); // Start with LED off

    // Warm-up delay for the MHZ19 sensor
    delay(2000);
}

void loop() {
    unsigned long currentMillis = millis();

    // Sensor Data Collection
    if (currentMillis - previousMillisMhz >= intervalMhz) {
        previousMillisMhz = currentMillis;

        digitalWrite(ledPin, HIGH); // Turn on the LED while collecting data
        Serial.print(F("Retrieving data... "));
        
        MHZ19_RESULT response = mhz.retrieveData();
        Serial.print(F("Response code: "));
        Serial.println(response);

        if (response == MHZ19_RESULT_OK) {
            Serial.print(F("CO2: "));
            Serial.println(mhz.getCO2());
            Serial.print(F("Temperature: "));
            Serial.println(mhz.getTemperature());
        } else {
            Serial.print(F("Error, code: "));
            Serial.println(response);
        }

        digitalWrite(ledPin, LOW); // Turn off the LED after collecting data
    }
}
`
What more can I do, to fix this or find out what the issue is?

strange-v, 2024-10-24:
Options:
1. In some framework versions SoftwareSerial was not reliable.
2. Wiring is not correct (e.g., RX from ESP should go to TX of the sensor)
3. The sensor is dead.

Error 5 means timeout in communication.

quintendewilde, 2024-10-24:
1. By framework you mean a firmware on the mhz19b? or something else. (In case I buy another I'm sure to buy the correct one. [I bought this one because it refers to your library](https://www.hobbyelectronica.nl/product/mh-z19b-co2-sensor/)
2. I've quadrouple checked it, it's correct.
3. See question one :)
Thanks for the feedback!

strange-v, 2024-10-24:
By framework, I mean Arduino/ESP-IDF.
I recommend trying any ESP32 (with HW serial) to ensure your sensor is okay.

quintendewilde, 2024-10-24:
Check, So if I test it in the same condition with my Lolin S3 mini (also Wemos) and get the same results, big chance that the sensor is dead.

quintendewilde, 2024-10-24:
I've tried with a Wemos S3 mini on port 43,44 for serial same error 5.

closed 2024-10-25 by quintendewilde